ARM-software/cmsis-pack-eclipse #104: Unable to see examples in the 'Examples' Tab on selecting a specific device

avi-77, 2022-10-17:
Hi @edriouk,

In the CMSIS pack manager plugin version 2.8.0, I'm unable to see the examples in the 'Examples' Tab on selecting a specific device. 
[image]
Although, I'm able to see the examples in the 'Examples' Tab on selecting All Devices as shown in the figure below.
[image]

Please note, I have installed the com.arm.cmsis.pack.refclient plugin. It was working fine in version 2.7.0. Could you please look into this issue for the next release?

edriouk, 2022-10-18:
Hi, 
that is a problem of comparing board IDs 
Board description:
`<board name="RSL10 Evaluation Board" revision="1.0" vendor="ONSemiconductor">`
Board ID: 
`board description is ONSemiconductor::RSL10 Evaluation Board (1.0)`
example board description:
`<board name="RSL10 Evaluation Board" vendor="ONSemiconductor"/>`
example board ID :
`ONSemiconductor::RSL10 Evaluation Board`

Boards of different revisions are treated as incompatible => check fails for a selected board/device on that board
This rule has been enforced in 2.8.0 because of adding Board Support functionality.
I agree that it can be made less restrictive to display available examples. 
Will be planned for the next release
Evgueni

andrewmclachlanadi, 2023-03-25:
Seeing this issue too. It's a noticable difference from previous releases and therefore looks like a severe bug.

edriouk, 2023-07-07:
Fixed in 2.9.0

closed 2023-07-07 by edriouk