ARM-software/cmsis-pack-eclipse #105: Error message shown when importing a cmsis project

AncaOprea11, 2023-05-15:
Hi,

An error message is shown when importing a cmsis project. This error looks like this:
[image]

!ENTRY com.arm.cmsis.pack.ui 4 0 2023-05-12 17:40:51.446
!MESSAGE Error updating RTE project
!STACK 1
org.eclipse.core.internal.resources.ResourceException(/mimi/mimi.rteconfig)[368]: java.lang.Exception: Resource '/mimi/mimi.rteconfig' does not exist.
	at org.eclipse.core.internal.resources.ResourceException.provideStackTrace(ResourceException.java:42)
	at org.eclipse.core.internal.resources.ResourceException.<init>(ResourceException.java:38)
	at org.eclipse.core.internal.resources.Resource.checkExists(Resource.java:330)
	at org.eclipse.core.internal.resources.Resource.checkAccessible(Resource.java:204)
	at org.eclipse.core.internal.resources.Resource.deleteMarkers(Resource.java:810)
	at com.arm.cmsis.pack.project.RteProjectUpdater.collectErrors(RteProjectUpdater.java:289)
	at com.arm.cmsis.pack.project.RteProjectUpdater.loadConfigFile(RteProjectUpdater.java:328)
	at com.arm.cmsis.pack.project.RteProjectUpdater.runInWorkspace(RteProjectUpdater.java:177)
	at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:43)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
!SUBENTRY 1 org.eclipse.core.resources 4 368 2023-05-12 17:40:51.446
!MESSAGE Resource '/mimi/mimi.rteconfig' does not exist.
!STACK 0
java.lang.Exception: Resource '/mimi/mimi.rteconfig' does not exist.
	at org.eclipse.core.internal.resources.ResourceException.provideStackTrace(ResourceException.java:42)
	at org.eclipse.core.internal.resources.ResourceException.<init>(ResourceException.java:38)
	at org.eclipse.core.internal.resources.Resource.checkExists(Resource.java:330)
	at org.eclipse.core.internal.resources.Resource.checkAccessible(Resource.java:204)
	at org.eclipse.core.internal.resources.Resource.deleteMarkers(Resource.java:810)
	at com.arm.cmsis.pack.project.RteProjectUpdater.collectErrors(RteProjectUpdater.java:289)
	at com.arm.cmsis.pack.project.RteProjectUpdater.loadConfigFile(RteProjectUpdater.java:328)
	at com.arm.cmsis.pack.project.RteProjectUpdater.runInWorkspace(RteProjectUpdater.java:177)
	at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:43)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

The .rteconfig exists physically on the disk and it is added into the project at import. 

The error is reproducible on MCUXpresso IDE 11.7.0 (eclipse 2022-06, cmsis arm version 2.7.0), we also tried an update to eclipse 2022-12, cmsis arm version 2.8.0 and is still reproducible, plus we observed the error on MDK 2020111904.

Steps to reproduce:
1. Create a CMSIS C/C++ Project for a device.
2. Export it to File System.
3. Delete the project from workspace or create a new workspace.
4. Import the project without copy the content in the workspace.

edriouk, 2024-03-10:
I cannot reproduce the problem in a vanilla Eclipse. However, I suspect that is RteProjectUpdater operations got called before the project is fully initialized. There are many participants that trigger the updater, including UI and CDT events.
The plug-ins already implement some measures to postpone the update before C-Project gets initialized, but obviously a not always.

The problem seems to be similar to one described here many years ago: 
https://www.eclipse.org/forums/index.php/t/95561/
Since even then there are no hooks nor events in Eclipse that can be used to react on import procedure.

edriouk, 2024-03-10:
@AncaOprea11 , I have noticed a strange thing in the stack trace: if a mimi.rteconfig would not exist, it the exception would be thrown prior to com.arm.cmsis.pack.project.RteProjectUpdater.collectErrors() call. Could you please have look into .cproject file to check if `mimi.rteconfig` is specified in .cproject file under `<rteConfig>` element?  Some older projects allowed to use a name that differs from project name.