ARM-software/cmsis-pack-eclipse #115: Error compiling startup_ARMCA7.c

SteMMo33, 2024-07-26:
Hi all,
I installed the last release: CmsisPackPlugIn2.10.0.zip 
I configured for my ST32MP135 board a simple blink code that uses ARMA/ as device and RTOS.
So I selected the following modules:

- CORE
- RTOS2 / Keil RTX5
- IRQ controller
- OS Tick API
- startup

When I run the build, after some succesful operation, I have the following error:

`Building file: ../RTE/Device/ARMCA7/startup_ARMCA7.c
Invoking: GNU Arm Cross C Compiler
arm-none-eabi-gcc -mcpu=cortex-a7 -mthumb -mlittle-endian -mfloat-abi=hard -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -g3 -D_RTE_ -DARMCA7 -I"C:/Users/stefano.mora/AppData/Local/Arm/Packs/ARM/CMSIS/5.6.0/CMSIS/Core_A/Include" -I"C:/Users/stefano.mora/AppData/Local/Arm/Packs/ARM/CMSIS/5.6.0/CMSIS/RTOS2/Include" -I"C:/Users/stefano.mora/AppData/Local/Arm/Packs/ARM/CMSIS/5.6.0/CMSIS/RTOS2/RTX/Include" -I"C:/Users/stefano.mora/AppData/Local/Arm/Packs/ARM/CMSIS/5.6.0/Device/ARM/ARMCA7/Include" -I"C:\Progetti\dbTechnologies\testSTM32\test_CMSIS_RTE/RTE" -I"C:\Progetti\dbTechnologies\testSTM32\test_CMSIS_RTE/RTE/CMSIS" -I"C:\Progetti\dbTechnologies\testSTM32\test_CMSIS_RTE/RTE/Device/ARMCA7" -isystem"C:/Users/stefano.mora/AppData/Local/Arm/Packs/ARM/CMSIS/5.6.0/CMSIS/Core_A/Include" -isystem"C:/Users/stefano.mora/AppData/Local/Arm/Packs/ARM/CMSIS/5.6.0/CMSIS/RTOS2/Include" -isystem"C:/Users/stefano.mora/AppData/Local/Arm/Packs/ARM/CMSIS/5.6.0/CMSIS/RTOS2/RTX/Include" -isystem"C:/Users/stefano.mora/AppData/Local/Arm/Packs/ARM/CMSIS/5.6.0/Device/ARM/ARMCA7/Include" -isystem"C:\Progetti\dbTechnologies\testSTM32\test_CMSIS_RTE/RTE" -isystem"C:\Progetti\dbTechnologies\testSTM32\test_CMSIS_RTE/RTE/CMSIS" -isystem"C:\Progetti\dbTechnologies\testSTM32\test_CMSIS_RTE/RTE/Device/ARMCA7" -std=gnu11 -MMD -MP -MF"RTE/Device/ARMCA7/startup_ARMCA7.d" -MT"RTE/Device/ARMCA7/startup_ARMCA7.o" -c -o "RTE/Device/ARMCA7/startup_ARMCA7.o" "../RTE/Device/ARMCA7/startup_ARMCA7.c"
C:\Users\STEFAN~1.MOR\AppData\Local\Temp\ccdvJ4Mp.s: Assembler messages:
C:\Users\STEFAN~1.MOR\AppData\Local\Temp\ccdvJ4Mp.s:78: Error: thumb conditional instruction should be in IT block -- `wfine '
make: *** [RTE/Device/ARMCA7/subdir.mk:39: RTE/Device/ARMCA7/startup_ARMCA7.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.
`

Any idea?

jkrech, 2024-07-29:
This does not ring a bell with me. Is this the first time you are trying this, or did this work correctly with previous versions of the plug-in? 
I have to admit that I am puzzled by the fact that you are writing that your application  is targeting "ST32MP135", however the selected device is a generic "ARMCA7" device. I do not expect this code to run on an STM32MP135 this way.

I do not see a CMSIS-Pack for STM32MP135 and you should probably contact  STMicroelectronics about this.

I would recommend using STM32CubeMX for creating bare metal projects in combination with the STM32CubeIDE.

SteMMo33, 2024-07-29:
Thanks a lot Joachim,
I'm doing a research on development tools for STM32MP devices.
I selected ARMCA7 precisely because I couldn't find the correct and most precise target.
So You confirm me that the MP135 target is not supported.

JonatanAntoni, 2024-07-30:
Hi @SteMMo33, Did you try to compile in regular ARM mode instead of THUMB?

Then, it looks like you are working with fairly old code based on CMSIS 5.6.0. With recent CMSIS 6 we moved the device support to https://github.com/ARM-software/Cortex_DFP/blob/main/Device/ARMCA7/Source/startup_ARMCA7.c

I'd recommend to update to latest code first. Next, you might need to tweak the startup code yourself because generic Cortex-A7 startup code does not consider device specific requirements for your ST32MP device.

SteMMo33, 2024-07-31:
Hi @JonatanAntoni ,
yes the CMSIS is version 5.5.1 but it is proposed by the tool as <latest>:
[image]

JonatanAntoni, 2024-07-31:
Don't get confused by pack vs component version. In your case its CMSIS 5.6.0 which ships RTX5 5.5.1. The latest versions available are CMSIS 6.1.0, CMSIS-RTX 5.9.0, and Cortex_DFP 1.1.0. With the major version jump to CMSIS 6 we splittet off RTX5 and device support into their own packs to reduce confusion and increase maintainability.

Open the CMSIS Pack perspective and update/install the new packs. Then the new component versions should show up in this RTE Configuraton dialog.